devicetree-org/dt-schema #18: linting tool for dt files

aparcar, 2019-07-31:
Hi, at OpenWrt there are sometimes badly styled DT files which waste human time to comment and fix, it would be nice to have some automatic checks for that. Is there a tool like [astyle](http://astyle.sourceforge.net/) or [clang-format](https://www.kernel.org/doc/html/latest/process/clang-format.html) for device tree files?

robherring, 2019-07-31:
I don't think any such thing exists for .dts files. Something like that should probably live within dtc, not this project, as it could re-use the parsing logic.

We need something too now for DT schema files. There's tools/yaml-format in this repo that mainly just does a round trip thru ruamel, but it doesn't quite format how I'd like it to. There's also yamllint, but I haven't looked at it.

aparcar, 2019-07-31:
With dtc I'm sure you refer to [this](https://github.com/dgibson/dtc) tool right? It supports `dts` as an output, however not linting. At OpenWrt [many](https://git.openwrt.org/?p=openwrt/openwrt.git;a=tree;f=target/linux/ath79/dts;h=71e6b5bc8a9064be46ba3442a2f97c4acf493460;hb=HEAD) `dts` files contain comments, so converting back and forth for linting isn't feasible. 

However, should I close this issue and reopen it at `dtc`?

robherring, 2019-07-31:
That's a mirror. The "official" repo is: https://git.kernel.org/pub/scm/utils/dtc/dtc.git

I didn't say dtc supported linting. It can output .dts files, but you loose a lot more than comments. 

I don't mind leaving the issue open here to track it. There's no issue tracking for dtc. You might want to raise the issue on devicetree-compiler@vger.kernel.org list.

aparcar, 2019-07-31:
Thanks, I sent a mail asking the same question. I'll update this issue once something happens