devicetree-org/dt-schema #27: setup.py inconsistencies and dependency issue

afaerber, 2019-10-26:
**Version:** There is a tag `v0.1` but both that tag's and master's `setup.py` specify it as `0.0.1`.
Could you please bump it to `0.2.0` and push some new `v0.2` tag? `README.md` makes it sound like one could do an installation into `.local`, but `pip` actually tries to install stuff to global `/usr`. Thus a saner way to create a distro package than `0.0.1~git<date-and-or-commit>` would be appreciated, now that the Linux kernel has started using it as dependency.

**URL:** The Linux kernel's `writing-schema.rst` points to this `devicetree-org/dt-schema.git` repo, but `setup.py` says `https://github.com/robherring/yaml-bindings` (which redirects to `https://github.com/robherring/dt-schema`). Which one is the official one to use?

**Tests:** Can `test/test-dt-validate.py` be integrated into `setup.py` so that `setup.py test` just works and one does not need to call that file explicitly? (Makefile doesn't work well in a multi-version Python build environment such as OBS.)

**Dependencies:** Please don't hardcode dependency versions. `jsonschema` 3.0.2 appears to work just fine with latest master, so can you please change from `jsonschema==3.0.1` to `jsonschema>=3.0.1` to allow installation on distros with newer packages?

robherring, 2019-11-07:
Thanks for the review. My first python project...

On my system (ubuntu), pip as root installs to /usr, but without root goes in the user's home dir.

While I'm happy to make packaging easier, there's a couple things you should be aware of. The only thing really tested for sure is current master of dt-schema with current Linus master and linux-next. I barely keep up with that. The expectation is users stay up to date with dt-schema. Current users are developers tracking current kernels anyways. Older kernels with current dt-schema will for sure have failures because the meta-schema checks get better (mostly implementing common mistakes in reviews as meta-schema) and I fix the existing schema for those new checks as we go. I generally keep any meta-schema changes that require kernel fixes in my local tree for a bit before publishing. Current kernels with older dt-schema will usually work, but then we miss the meta-schema improvements. There's a variety of ways to solve these issues when necessary. I just hope to put that off, and don't want distro packaging to be the reason.

robherring, 2019-11-07:
@afaerber PR #28 fixes this.

FYI, I'm going to switch to date based versions. YYYY.MM

closed 2020-01-23 by robherring