devicetree-org/dt-schema #35: License and PyPi release

sorki, 2020-04-26:
Hi,
the license of the project is not clear (would be nice if it had a `LICENSE` file). 

If you publish this on PyPi it would make packaging easier as well.

robherring, 2020-04-27:
Every file has an SPDX tag.

sorki, 2020-04-28:
Oh, never mind then, PyPi release would be nice though.

robherring, 2020-04-29:
Yeah, PyPi releases are on my todo. I've delayed it as the project has been evolving rather quickly and you pretty much need to be using the tip. But hopefully things are starting to mature a bit.

sorki, 2020-04-29:
No problem, take your time, I will use the GitHub repo directly meanwhile and update when available.

Since the package builds correctly with `setup.py` it should be just a matter of `python setup.py sdist upload` and proper configuration of PyPi access (and registration).

robherring, 2020-06-19:
Uploaded to PyPI: https://pypi.org/project/dtschema/2020.6rc1/

Should resolve #41 too.

closed 2020-08-12 by robherring