devicetree-org/dt-schema #41: v2020.05 tarball does not build

afaerber, 2020-06-17:
`setup.py` (`setuptools_scm`) appears to be looking for some version metadata that the released GitHub `dt-schema-2020.05.tar.gz` tarball from the Releases tab does not contain:

```
[   30s] + /usr/bin/python3 setup.py build '--executable=/usr/bin/python3 -s'
[   30s] Traceback (most recent call last):
[   30s]   File "setup.py", line 20, in <module>
[   30s]     setuptools.setup(
[   30s]   File "/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 145, in setup
[   30s]     return distutils.core.setup(**attrs)
[   30s]   File "/usr/lib/python3.8/distutils/core.py", line 108, in setup
[   30s]     _setup_distribution = dist = klass(attrs)
[   30s]   File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 447, in __init__
[   30s]     _Distribution.__init__(self, {
[   30s]   File "/usr/lib/python3.8/distutils/dist.py", line 292, in __init__
[   30s]     self.finalize_options()
[   30s]   File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 740, in finalize_options
[   30s]     ep.load()(self)
[   31s]   File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 747, in _finalize_setup_keywords
[   31s]     ep.load()(self, ep.name, value)
[   31s]   File "/usr/lib/python3.8/site-packages/setuptools_scm/integration.py", line 17, in version_keyword
[   31s]     dist.metadata.version = _get_version(config)
[   31s]   File "/usr/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 147, in _get_version
[   31s]     parsed_version = _do_parse(config)
[   31s]   File "/usr/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 110, in _do_parse
[   31s]     raise LookupError(
[   31s] LookupError: setuptools-scm was unable to detect version for '/home/abuild/rpmbuild/BUILD/dt-schema-2020.05'.
[   31s] 
[   31s] Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
[   31s] 
[   31s] For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
```

On the other hand, #35 indicates that a PyPI tarball is not available yet, seemingly leaving us with no buildable tarball. Since the tarball is not just from some tag but also announced under Releases, I would expect it to be usable as is.

By `sed -i "s|'setuptools_scm'||" setup.py` and `echo '__version__ = "2020.5"' > dtschema/version.py` I can get the build to finish.

robherring, 2020-06-18:
I guess generating the version from git tag will never work unless PyPI somehow does this.

afaerber, 2020-06-19:
The GitHub tarball has a subdir `dt-schema-2020.05` that you could extract the `2020.05` from, as fallback, if you don't insist on it reading `2020.5` (although stripping the zero could probably be done by numeric parsing and reformatting, if desired).

closed 2020-08-12 by robherring