devicetree-org/dt-schema #49: dt-validate: TypeError: 'str' object does not support item assignment

xypron, 2020-11-03:
I try to validate a device-tree in yaml format.
[a.yaml.txt](https://github.com/devicetree-org/dt-schema/files/5480654/a.yaml.txt)

Unfortunately this results in an error:

```
$ dt-validate /tmp/a.yaml 
Traceback (most recent call last):
  File "./dt-validate", line 159, in <module>
    sg.check_trees(filename, testtree)
  File "./dt-validate", line 108, in check_trees
    self.check_subtree(dt, subtree, "/", "/", filename)
  File "./dt-validate", line 94, in check_subtree
    self.check_node(tree, subtree, nodename, fullname, filename)
  File "./dt-validate", line 39, in check_node
    node['$nodename'] = [ nodename ]
TypeError: 'str' object does not support item assignment
```

robherring, 2020-11-03:
The yaml encoding supported by dtschema is the yaml output from dtc. That's not the same as dts2yaml.

closed 2020-11-03 by robherring

xypron, 2020-11-03:
We should mention that the devicetree compiler can be used for dts to yaml transcoding in the README.md. Unfortunately even current Linux distributions (e.g. Debian Bullseye) come with dtc versions before 1.5.0 and therefore this is not evident.

xypron, 2020-11-03:
There seems to be a bug in dtc. It generates:

    compatible: ["wand,imx6q-wandboard\0fsl,imx6q"]

instead of

    compatible: ["wand,imx6q-wandboard", "fsl,imx6q"]

resulting in a dt-validate warning:

    compatible:0: 'wand,imx6q-wandboard\x00fsl,imx6q' does not match '^[a-zA-Z][a-zA-Z0-9,+\\-._]+$'

robherring, 2020-11-03:
Are you doing dtb->yaml or dtb->dts->yaml as that wouldn't work. Only dts->yaml works as there is some type information in dts files that you lose in a dtb.

robherring, 2020-11-03:
It appears to be fine here: https://gitlab.com/robherring/linux-dt-bindings/-/jobs/826489577