devicetree-org/dt-schema #55: Traceback when naively trying to check a dt example in yaml format

schlatterbeck, 2021-05-19:
I'm trying to contribute some devicetree documentation and trying to verify my example. May naive usage:
```
dt-validate Documentation/devicetree/bindings/auxdisplay/hit,hd44780+nxp,pcf8575.yaml
```
produces:
```
Traceback (most recent call last):
  File "/usr/local/bin/dt-validate", line 164, in <module>
    sg.check_trees(filename, testtree)
  File "/usr/local/bin/dt-validate", line 113, in check_trees
    self.check_subtree(dt, subtree, "/", "/", filename)
  File "/usr/local/bin/dt-validate", line 99, in check_subtree
    self.check_node(tree, subtree, nodename, fullname, filename)
  File "/usr/local/bin/dt-validate", line 39, in check_node
    node['$nodename'] = [ nodename ]
TypeError: 'str' object does not support item assignment
```

Now I'm quite sure this is an error in my usage of the tool (but the tool should help me with a message and not throw a traceback). Can you give me (or preferrably add this to the documentation on this page) a quick start how to verify a single yaml file against the schemata in-kernel?
I'm trying to get some documentation into the kernel and I'm at the point where it produces a warning that I cannot reproduce locally.

robherring, 2021-05-20:
In the kernel tree, set DT_SCHEMA_FILES:

make DT_SCHEMA_FILES=Documentation/devicetree/bindings/auxdisplay/hit,hd44780+nxp,pcf8575.yaml dt_binding_check

It's documented in Documentation/devicetree/bindings/writing-schema.rst

robherring, 2021-06-04:
Also, 'dt-validate' is for YAML encoded DT files as output from dtc. 'dt-doc-validate' is for validating the schema(aka binding) files which is what you have above.

closed 2021-06-04 by robherring