devicetree-org/dt-schema #59: dt-doc-validate crashes when validating (probably incorrect) YAML file

okias, 2021-09-11:
```
$ dt-doc-validate -u Documentation/devicetree/bindings Documentation/devicetree/bindings/i2c/nvidia,tegra-i2c.yaml
Traceback (most recent call last):
  File "/home/okias/.local/bin/dt-doc-validate", line 67, in <module>
    ret = check_doc(f)
  File "/home/okias/.local/bin/dt-doc-validate", line 33, in check_doc
    for error in sorted(dtschema.DTValidator.iter_schema_errors(testtree), key=lambda e: e.linecol):
  File "/home/okias/.local/lib/python3.9/site-packages/dtschema/lib.py", line 724, in iter_schema_errors
    cls(meta_schema).annotate_error(error, meta_schema, error.schema_path)
  File "/home/okias/.local/lib/python3.9/site-packages/dtschema/lib.py", line 707, in annotate_error
    schema = schema[p]
KeyError: 'properties'
```
content of `nvidia,tegra-i2c.yaml`
```
# SPDX-License-Identifier: (GPL-2.0-only OR BSD-2-Clause)
%YAML 1.2
---
$id: http://devicetree.org/schemas/i2c/nvidia,tegra-i2c.yaml#
$schema: http://devicetree.org/meta-schemas/core.yaml#

title: NVIDIA Tegra I2C controller driver

maintainers:
  - No-one <no-one@no-one.org>

allOf:
  - $ref: /schemas/i2c/i2c-controller.yaml#
  - if:
      properties:
        compatible:
          contains:
            enum:
              - nvidia,tegra20-i2c
              - nvidia,tegra30-i2c
    then:
      properties:
        clock-names:
          items:
            - const: div-clk
            - const: fast-clk
  - if:
      properties:
        compatible:
          contains:
            enum:
              - nvidia,tegra114-i2c
              - nvidia,tegra210-i2c
    then:
      properties:
        clock-names:
          items:
            - const: div-clk
  - if:
      properties:
        compatible:
          contains:
            - const: nvidia,tegra210-i2c-vi
    then:
      properties:
        clock-names:
          items:
            - const: div-clk
            - const: slow

properties:
  compatible:
    oneOf:
      - description: |
          Tegra20 has 4 generic I2C controller. This can support
          master and slave mode of I2C communication. The i2c-tegra driver
          only support master mode of I2C communication.
        items:
          - const: nvidia,tegra20-i2c

      - description: |
          Tegra20 has specific I2C controller called as DVC I2C
          controller. This only support master mode of I2C communication.
          Register interface/offset and interrupts handling are different
          than generic I2C controller.
        items:
          - const: nvidia,tegra20-i2c-dvc

      - description: |
          Tegra30 has 5 generic I2C controller. This controller is
          very much similar to Tegra20 I2C controller with additional feature:
          Continue Transfer Support. This feature helps to implement M_NO_START
          as per I2C core API transfer flags. Driver of I2C controller is
          compatible with "nvidia,tegra30-i2c" to enable the continue transfer
          support. This is also compatible with "nvidia,tegra20-i2c" without
          continue transfer support.
        items:
          - const: nvidia,tegra30-i2c
          - const: nvidia,tegra20-i2c

      - description: |
          Tegra114 has 5 generic I2C controller. This controller is very
          much similar to Tegra30 I2C controller with some hardware modification:
          - Tegra30/Tegra20 I2C controller has 2 clock source called div-clk and
          fast-clk. Tegra114 has only one clock source called as div-clk and
          hence clock mechanism is changed in I2C controller.
          - Tegra30/Tegra20 I2C controller has enabled per packet transfer by
          default and there is no way to disable it. Tegra114 has this
          interrupt disable by default and SW need to enable explicitly.
          Due to above changes, Tegra114 I2C driver makes incompatible with
          previous hardware driver.
        items:
          - const: nvidia,tegra114-i2c

      - items:
          - const: nvidia,tegra124-i2c
          - const: nvidia,tegra114-i2c
      - items:
          - const: nvidia,tegra210-i2c
          - const: nvidia,tegra124-i2c
      - description: |
          Tegra210 has one I2C controller that is on host1x bus
          and is part of VE power domain and typically used for camera use-cases.
          This VI I2C controller is mostly compatible with the programming model
          of the regular I2C controllers with a few exceptions. The I2C registers
          start at an offset of 0xc00 (instead of 0), registers are 16 bytes
          apart (rather than 4) and the controller does not support slave mode.
        items:
          - const: nvidia,tegra210-i2c-vi
      - items:
          - const: nvidia,tegra186-i2c
          - const: nvidia,tegra210-i2c

  reg: true
  interrupts: true

  '#address-cells':
    const: 1
  '#size-cells':
    const: 0

  clocks: true
  clock-names: true

  resets: true
  reset-names:
    items:
      - i2c

  power-domains:
    description: |
      Only for nvidia,tegra210-i2c-vi compatible node and must
      include venc powergate node as vi i2c is part of VE power domain.
      tegra210-i2c-vi:
        - pd_venc
  dmas:
    description: |
      Must contain an entry for each entry in clock-names.
  dma-names:
    items:
      - rx
      - tx

required:
  - compatible
  - clocks
  - clock-names

additionalProperties: false

examples:
  - |
    i2c@7000c000 {
      compatible = "nvidia,tegra20-i2c";
      reg = <0x7000c000 0x100>;
      interrupts = <0 38 0x04>;
      #address-cells = <1>;
      #size-cells = <0>;
      clocks = <&tegra_car 12>, <&tegra_car 124>;
      clock-names = "div-clk", "fast-clk";
      resets = <&tegra_car 12>;
      reset-names = "i2c";
      dmas = <&apbdma 16>, <&apbdma 16>;
      dma-names = "rx", "tx";
    };
```

I'm pretty sure the YAML file is wrong, just reporting the `dt-doc-validate` crash

okias, 2021-09-11:
ok, it was line
```
            - const: nvidia,tegra210-i2c-vi
```
which has to be changed to
```
            const: nvidia,tegra210-i2c-vi
```

robherring, 2021-09-13:
On Sat, Sep 11, 2021 at 1:53 PM David Heidelberg
***@***.***> wrote:
>
> ok, it was line
>
>             - const: nvidia,tegra210-i2c-vi
>
> which has to be changed to
>
>             const: nvidia,tegra210-i2c-vi

Thanks for the report. I actually just ran into this too. The problem
is in walking the schema path when the schema has an 'if' with an
error in. I have a fix I need to push out.

robherring, 2021-10-05:
Fixed with commit c5211d872a0906e1

closed 2021-10-05 by robherring