devicetree-org/dt-schema #67: dt-doc-validate won't check blocks inside allOf: - if:

okias, 2022-01-08:
Ignores all problems inside structures as:

```
allOf:
  - if:
    ....
    then:
      _CODE_PLACED_HERE_ISN'T_CHECKED
```

robherring, 2022-01-08:
That is at least partially on purpose. Schemas in an if/then schema are generally incomplete as they are further restrictions on the top-level schema. We should be checking against at least the json-schema meta-schema, but that is not very strict (as the spec says any unknown vocabulary (such as a typo) is silently ignored). We should be at least checking for valid keywords (defined in meta-schemas/keywords.yaml). Can you confirm?

okias, 2022-01-08:
I got issues with missing `description` for `properties` and `patternProperties` which lead to (probably) ignoring my entries. Cannot say much more about it

robherring, 2022-01-10:
The check for a json-schema keyword used as a DT property name was not being checked in all cases. This is a common error in the schema structure. Now fixed in main.

closed 2022-01-10 by robherring