devicetree-org/dt-schema #68: Fail to validate DT with jsonschema>3

montjoie, 2022-01-27:
On my gentoo dt checks works with jsonschema-3.2.0
But after upgrading on 4.3.3 I got:
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.9/dt-mk-schema", line 38, in <module>
    schemas = dtschema.process_schemas(args.schemas, core_schema=(not args.useronly))
  File "/usr/lib/python3.9/site-packages/dtschema/lib.py", line 587, in process_schemas
    sch = process_schema(os.path.abspath(filename))
  File "/usr/lib/python3.9/site-packages/dtschema/lib.py", line 568, in process_schema
    DTValidator.check_schema(schema)
  File "/usr/lib/python3.9/site-packages/dtschema/lib.py", line 699, in check_schema
    for error in cls(meta_schema).iter_errors(schema):
  File "/usr/lib/python3.9/site-packages/dtschema/lib.py", line 681, in __init__
    jsonschema.Draft7Validator.__init__(self, schema, types, resolver=self.resolver,
TypeError: __init__() got multiple values for argument 'resolver'
make[2]: *** [/usr/src/linux-next/Documentation/devicetree/bindings/Makefile:76 : Documentation/devicetree/bindings/processed-schema-examples.json] Erreur 1
make[2]: *** Suppression du fichier « Documentation/devicetree/bindings/processed-schema-examples.json »
make[1]: *** [/usr/src/linux-next/Makefile:1471 : dt_binding_check] Erreur 2

robherring, 2022-01-27:
You need to update dtschema. jsonschema 4.x.x broke us such as the above. More importantly, newer versions support 'unevaluatedProperties' which we need.

closed 2022-01-27 by robherring

montjoie, 2022-01-27:
I already use latest dtschema 2021.12

robherring, 2022-01-27:
You aren't because the above issue was fixed in 2021.10 and Draft7Validator is not used in 2021.12. 2021.12 switched to Draft 2019-09 and also requires json-schema 4.x.x.

This is the fix for your issue:
commit ae00cf651d34bf822f64f310e3fc2a4255c7a4e1
Author: Rob Herring <robh@kernel.org>
Date:   Thu Sep 30 08:39:29 2021 -0500

    dtschema: Rework __init__ arg handling for jsonschema 4.0.0
    
    jsonschema 4.0.0 changed how Validator class attrs are initialized which
    broke us. Update how we pass keyword args.
    
    Sub-classing a validator is not supported according to the jsonschema
    author and subject to breaking (as it just did), so we will need to
    rework the interface.
    
    Signed-off-by: Rob Herring <robh@kernel.org>

montjoie, 2022-01-27:
I am sorry, in fact dt-schema was not well upgraded. Now I am really on 2021.12 and it works. Sorry for the error