devicetree-org/dt-schema #70: Old example YAML files break dt-doc-validate / dt-mk-schema

rmilecki, 2022-03-30:
Before updating to the latest `dtschema` I was using 2021.2 or 2021.3 release to work with in-Linux YAML files. That left me with some `foo.example.dt.yaml` files.

After switching to the latest Linux sources & updating `dtschema` Linux's `dtbs_check` stopped working for me without printing any meaningful error. I had to install `dt-schema` locally and debug it.

## Breaking `dt-doc-validate`

In my Linux sources directory I have following file:
```
> cat Documentation/devicetree/bindings/pinctrl/qcom,mdm9607-pinctrl.example.dt.yaml
---
- compatible: ["foo"]
  model: ["foo"]
  interrupt-parent: [[!phandle 0xffffffff]]
  '#address-cells': [[0x1]]
  '#size-cells': [[0x1]]
  example-0:
    '#address-cells': [[0x1]]
    '#size-cells': [[0x1]]
    pinctrl@1000000:
      compatible: ["qcom,mdm9607-tlmm"]
      reg: [[0x1000000, 0x300000]]
      interrupts: [[0x0, 0xd0, 0x4]]
      gpio-controller: true
      gpio-ranges: [[!phandle 0xffffffff, 0x0, 0x0, 0x50]]
      '#gpio-cells': [[0x2]]
      interrupt-controller: true
      '#interrupt-cells': [[0x2]]
  __fixups__:
    foo: ["/:interrupt-parent:0"]
    msmgpio: ["/example-0/pinctrl@1000000:gpio-ranges:0"]
...
```
It causes:
```
Traceback (most recent call last):
  File "/home/rmilecki/.local/bin/dt-doc-validate", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/rmilecki/dt-schema/tools/dt-doc-validate", line 71, in <module>
    ret = check_doc(f)
  File "/home/rmilecki/dt-schema/tools/dt-doc-validate", line 37, in check_doc
    for error in sorted(dtschema.DTValidator.iter_schema_errors(testtree), key=lambda e: e.linecol):
  File "/home/rmilecki/dt-schema/dtschema/lib.py", line 1019, in iter_schema_errors
    meta_schema = cls.resolver.resolve_from_url(schema['$schema'])
TypeError: list indices must be integers or slices, not str
```
It's because in case of above YAML file variable `schema` is not `dict` but a following list:
```
[{'compatible': ['foo'], 'model': ['foo'], 'interrupt-parent': [[4294967295]], '#address-cells': [[1]], '#size-cells': [[1]], 'example-0': {'#address-cells': [[1]], '#size-cells': [[1]], 'pinctrl@1000000': {'compatible': ['qcom,mdm9607-tlmm'], 'reg': [[16777216, 3145728]], 'interrupts': [[0, 208, 4]], 'gpio-controller': True, 'gpio-ranges': [[4294967295, 0, 0, 80]], '#gpio-cells': [[2]], 'interrupt-controller': True, '#interrupt-cells': [[2]]}}, '__fixups__': {'foo': ['/:interrupt-parent:0'], 'msmgpio': ['/example-0/pinctrl@1000000:gpio-ranges:0']}}]
```

## Breaking dt-mk-schema
In my Linux sources directory I have following file:
```
> cat Documentation/devicetree/bindings/mtd/partitions/brcm,trx.example.dt.yaml
---
- compatible: ["foo"]
  model: ["foo"]
  interrupt-parent: [[!phandle 0xffffffff]]
  '#address-cells': [[0x1]]
  '#size-cells': [[0x1]]
  example-0:
    '#address-cells': [[0x1]]
    '#size-cells': [[0x1]]
    partitions:
      compatible: ["brcm,trx"]
  __fixups__:
    foo: ["/:interrupt-parent:0"]
...
```
It causes:
```
Traceback (most recent call last):
  File "/home/rmilecki/.local/bin/dt-mk-schema", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/rmilecki/dt-schema/tools/dt-mk-schema", line 38, in <module>
    schemas = dtschema.process_schemas(args.schemas, core_schema=(not args.useronly))
  File "/home/rmilecki/dt-schema/dtschema/lib.py", line 719, in process_schemas
    sch = process_schema(os.path.abspath(filename))
  File "/home/rmilecki/dt-schema/dtschema/lib.py", line 695, in process_schema
    DTValidator.check_schema(schema)
  File "/home/rmilecki/dt-schema/dtschema/lib.py", line 1036, in check_schema
    meta_schema = cls.resolver.resolve_from_url(schema['$schema'])
TypeError: list indices must be integers or slices, not str
make[1]: *** [Documentation/devicetree/bindings/Makefile:70: Documentation/devicetree/bindings/processed-schema.json] Error 1
```
It's because in case of above YAML file variable `schema` is not `dict` but a following list:
```
[{'compatible': ['foo'], 'model': ['foo'], 'interrupt-parent': [[4294967295]], '#address-cells': [[1]], '#size-cells': [[1]], 'example-0': {'#address-cells': [[1]], '#size-cells': [[1]], 'partitions': {'compatible': ['brcm,trx']}}, '__fixups__': {'foo': ['/:interrupt-parent:0']}}]
```

rmilecki, 2022-03-30:
I've fixed that by
```
find ./Documentation/ -name "*.example.dt.yaml" -exec rm {} \;
```
but I think `lib.py` should be improved not to crash & to display a meaningful error message.

robherring, 2022-03-31:
It's the kernel update to 5.18-rc that caused the issue as it stopped using .dt.yaml files. A 'make clean' will still remove them.

I will look at handling the error better.

robherring, 2022-03-31:
I fixed and made a new release as others are likely to hit this when rc1 is tagged.

closed 2022-03-31 by robherring