devicetree-org/dt-schema #87: TypeError: 'str' object does not support item assignment

sjg20, 2022-10-04:
I think the README could be updated with more concrete examples as I have forgotten how to use this. I get:

f2b7cbf (HEAD, us/master) dtschema: Relax validation of schemas when processed

$ tools/dt-mk-schema >asc
$ tools/dt-validate -s asc test/schemas/good-example.yaml 
Traceback (most recent call last):
  File "/scratch/sglass/cosarm/dt-schema/tools/dt-validate", line 175, in <module>
    sg.check_trees(filename, testtree)
  File "/scratch/sglass/cosarm/dt-schema/tools/dt-validate", line 122, in check_trees
    self.check_subtree(dt, subtree, False, "/", "/", filename)
  File "/scratch/sglass/cosarm/dt-schema/tools/dt-validate", line 106, in check_subtree
    self.check_node(tree, subtree, disabled, nodename, fullname, filename)
  File "/scratch/sglass/cosarm/dt-schema/tools/dt-validate", line 38, in check_node
    node['$nodename'] = [ nodename ]
TypeError: 'str' object does not support item assignment

robherring, 2022-10-05:
It could indeed use some love. Your issue is running dt-validate on a schema, not a DT. You need to use a dtb directly or convert a dts to yaml. Support for the latter is bit rotting as it isn't really being used now.

robherring, 2022-10-10:
@sjg20 here's some improvements: https://github.com/robherring/dt-schema/tree/readme

sjg20, 2022-11-02:
OK that looks better. It still got an error, per my patch earlier today.

Are you going to apply it to the other tree? If so I say close this.

closed 2022-11-02 by robherring