devicetree-org/dt-schema #98: TypeError: unhashable type: 'dict'

lumag, 2023-02-05:
With the dt-schema 2023.1 and Python 3.11 (Debian testing) I'm getting the following erorr now:

```
$ PATH=~/.local/bin/:$PATH make -C ../build-64/ ARCH=arm64 CROSS_COMPILE="ccache aarch64-linux-gnu-" dt_binding_check DT_SCHEMA_FILES=msm/gpu.yaml
make: Entering directory '/home/lumag/Projects/Qcomm/build-64'
  LINT    Documentation/devicetree/bindings
  CHKDT   Documentation/devicetree/bindings/processed-schema.json
  SCHEMA  Documentation/devicetree/bindings/processed-schema.json
Traceback (most recent call last):
  File "/home/lumag/.local/bin//dt-mk-schema", line 38, in <module>
    schemas = dtschema.set_schemas(args.schemas, core_schema=(not args.useronly))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 1087, in set_schemas
    schema_cache = process_schemas(schema_files, core_schema)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 735, in process_schemas
    sch = process_schema(os.path.abspath(filename))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 713, in process_schema
    DTValidator.check_schema(schema, strict=False)
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 1238, in check_schema
    fixup_schema(schema)
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 434, in fixup_schema
    add_select_schema(schema)
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 599, in add_select_schema
    compatible_list = extract_node_compatibles(schema['properties']['compatible'])
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lumag/.local/lib/python3.11/site-packages/dtschema/lib.py", line 531, in extract_node_compatibles
    compatible_list.update(l)
TypeError: unhashable type: 'dict'
make[1]: *** [/home/lumag/Projects/Qcomm/kernel/Documentation/devicetree/bindings/Makefile:68: Documentation/devicetree/bindings/processed-schema.json] Error 1
make[1]: *** Deleting file 'Documentation/devicetree/bindings/processed-schema.json'
make: *** [/home/lumag/Projects/Qcomm/kernel/Makefile:1511: dt_binding_check] Error 2
make: Leaving directory '/home/lumag/Projects/Qcomm/build-64'
```

lumag, 2023-02-05:
This was caused by a binding's patch by other developer, which has:
```yamlproperties:
  compatible:
    enum:
      - qcom,sc7180-dp
      - items:
          - enum:
              - qcom,sm8450-dp
              - qcom,sm8550-dp
          - const: qcom,sm8350-dp
```

So, most likely, this can now be closed unless you want to handle such complex enum entries (I don't even know if this is valid or not).

robherring, 2023-02-06:
Did this just start with python 3.11? I've been using it since before it was released. 

That's not a valid schema anyways. Looks like a 'oneOf' is needed in there.

closed 2023-02-06 by robherring