devicetree-org/dt-schema #112: KeyError: 'type'

sjg20, 2023-10-24:
I see this on Linux ommit:
I see this with Linux -next:

f017d9a92a73 (tag: next-20231024, next/master) Add linux-next specific files for 20231024

make dt_binding_check -s -j30 
/home/sglass/cosarm/linux/Documentation/devicetree/bindings/mfd/qcom-pm8xxx.yaml:
Error in referenced schema matching $id: http://devicetree.org/schemas/input/qcom,pm8921-keypad.yaml
Traceback (most recent call last):
  File "/home/sglass/.local/bin/dt-doc-validate", line 64, in <module>
    ret |= check_doc(f)
  File "/home/sglass/.local/bin/dt-doc-validate", line 32, in check_doc
    for error in sorted(dtsch.iter_errors(), key=lambda e: e.linecol):
  File "/home/sglass/.local/lib/python3.10/site-packages/dtschema/schema.py", line 125, in iter_errors
    self.annotate_error(scherr, meta_schema, scherr.schema_path)
  File "/home/sglass/.local/lib/python3.10/site-packages/dtschema/schema.py", line 104, in annotate_error
    schema = schema[p]
KeyError: 'type'
Traceback (most recent call last):
  File "/home/sglass/.local/bin/dt-doc-validate", line 64, in <module>
    ret |= check_doc(f)
  File "/home/sglass/.local/bin/dt-doc-validate", line 32, in check_doc
    for error in sorted(dtsch.iter_errors(), key=lambda e: e.linecol):
  File "/home/sglass/.local/lib/python3.10/site-packages/dtschema/schema.py", line 125, in iter_errors
    self.annotate_error(scherr, meta_schema, scherr.schema_path)
  File "/home/sglass/.local/lib/python3.10/site-packages/dtschema/schema.py", line 104, in annotate_error
    schema = schema[p]
KeyError: 'type'
/home/sglass/.local/lib/python3.10/site-packages/dtschema/schemas/reserved-memory/shared-dma-pool.yaml: warning: ignoring duplicate '$id' value 'http://devicetree.org/schemas/reserved-memory/shared-dma-pool.yaml#'
/home/sglass/.local/lib/python3.10/site-packages/dtschema/schemas/reserved-memory/memory-region.yaml: warning: ignoring duplicate '$id' value 'http://devicetree.org/schemas/reserved-memory/memory-region.yaml#'
/home/sglass/.local/lib/python3.10/site-packages/dtschema/schemas/reserved-memory/reserved-memory.yaml: warning: ignoring duplicate '$id' value 'http://devicetree.org/schemas/reserved-memory/reserved-memory.yaml#'
/home/sglass/.local/lib/python3.10/site-packages/dtschema/schemas/reserved-memory/framebuffer.yaml: warning: ignoring duplicate '$id' value 'http://devicetree.org/schemas/reserved-memory/framebuffer.yaml#'
/home/sglass/cosarm/linux/Documentation/devicetree/bindings/sound/renesas,rsnd.example.dtb: sound@ec500000: port:endpoint: Unevaluated properties are not allowed ('phandle' was unexpected)
	from schema $id: http://devicetree.org/schemas/sound/renesas,rsnd.yaml#
/home/sglass/cosarm/linux/Documentation/devicetree/bindings/sound/renesas,rsnd.example.dtb: sound@ec500000: Unevaluated properties are not allowed ('port' was unexpected)
	from schema $id: http://devicetree.org/schemas/sound/renesas,rsnd.yaml#

robherring, 2023-10-24:
Already fixed in commit d873ca7.

closed 2023-10-24 by robherring

sjg20, 2023-10-24:
Hmmm, I used this just an hour ago to install: