devicetree-org/dt-schema #113: How to install from git

sjg20, 2023-10-24:
As per the docs:

https://github.com/devicetree-org/dt-schema

it is possible to install the latest from github:

pip3 install git+https://github.com/devicetree-org/dt-schema.git@main

This is what I get:

Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/devicetree-org/dt-schema.git@main
  Cloning https://github.com/devicetree-org/dt-schema.git (to revision main) to /tmp/pip-req-build-60tnocl4
  Running command git clone --filter=blob:none --quiet https://github.com/devicetree-org/dt-schema.git /tmp/pip-req-build-60tnocl4
  Resolved https://github.com/devicetree-org/dt-schema.git to commit 411c305105dd127331f54c32a59e4e648b726138
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done

It does not seem to install anything, so (without noticing) I was still on the 'PyPI' version

sjg20, 2023-10-25:
For now I find I can use it like this:

PATH=$PATH:/home/sglass/cosarm/dt-schema/tools/ PYTHONPATH=/home/sglass/cosarm/dt-schema make DT_CHECKER_FLAGS=-m dt_binding_check -s -j30 DT_SCHEMA_FILES=mtd/partitions

robherring, 2023-10-25:
Humm, works for me(TM):

$ pip3 install --break-system-packages git+https://github.com/devicetree-org/dt-schema.git@main
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/devicetree-org/dt-schema.git@main
  Cloning https://github.com/devicetree-org/dt-schema.git (to revision main) to /tmp/pip-req-build-j66xqsin
  Running command git clone --filter=blob:none --quiet https://github.com/devicetree-org/dt-schema.git /tmp/pip-req-build-j66xqsin
  Resolved https://github.com/devicetree-org/dt-schema.git to commit 79243b1c3dd4f0c1e73c840ecd3148c72d736f72
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: ruamel.yaml>0.15.69 in /usr/lib/python3/dist-packages (from dtschema==2023.10.dev9+g79243b1c3dd4) (0.17.21)
Requirement already satisfied: jsonschema<4.18,>=4.1.2 in /home/rob/.local/lib/python3.11/site-packages (from dtschema==2023.10.dev9+g79243b1c3dd4) (4.17.3)
Requirement already satisfied: rfc3987 in /usr/lib/python3/dist-packages (from dtschema==2023.10.dev9+g79243b1c3dd4) (1.3.8)
Requirement already satisfied: pylibfdt in /home/rob/.local/lib/python3.11/site-packages (from dtschema==2023.10.dev9+g79243b1c3dd4) (1.7.0.post1)
Requirement already satisfied: attrs>=17.4.0 in /home/rob/.local/lib/python3.11/site-packages (from jsonschema<4.18,>=4.1.2->dtschema==2023.10.dev9+g79243b1c3dd4) (22.2.0)
Requirement already satisfied: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in /usr/lib/python3/dist-packages (from jsonschema<4.18,>=4.1.2->dtschema==2023.10.dev9+g79243b1c3dd4) (0.18.1)
Building wheels for collected packages: dtschema
  Building wheel for dtschema (pyproject.toml) ... done
  Created wheel for dtschema: filename=dtschema-2023.10.dev9+g79243b1c3dd4-py3-none-any.whl size=99845 sha256=571a008480ee4695fbd514c33d766eed856cc9b5b7b9c3aeebb996ba6348064a
  Stored in directory: /tmp/pip-ephem-wheel-cache-wb_5y6q1/wheels/34/5d/fa/f2d476ace200c637ffcc051d0083345e988949d9aa14e7560c
Successfully built dtschema
Installing collected packages: dtschema
  Attempting uninstall: dtschema
    Found existing installation: dtschema 2023.10.dev6+g1318b0345946
    Uninstalling dtschema-2023.10.dev6+g1318b0345946:
      Successfully uninstalled dtschema-2023.10.dev6+g1318b0345946
Successfully installed dtschema-2023.10.dev9+g79243b1c3dd4
$ dt-validate --version
2023.10.dev9+g79243b1c3dd4

robherring, 2023-10-25:
That was with an editable install of dtschema. I did an install from PyPI and that worked too.

closed 2024-01-02 by robherring

sjg20, 2024-01-02:
Yes, that seems to be it, since it works now. Thanks