devicetree-org/dt-schema #132: Missing stdin-path in chosen node schema

vstehle, 2024-03-01:
The `/chosen` node [YAML schema](https://github.com/devicetree-org/dt-schema/blob/main/dtschema/schemas/chosen.yaml) is missing a property for `stdin-path`, while [it is defined](https://devicetree-specification.readthedocs.io/en/latest/chapter3-devicenodes.html?highlight=chosen#chosen-node) in the Devicetree specification.

The two other `/chosen` node properties defined in the specification, `bootargs` and `stdout-path`, do indeed have a corresponding property in the YAML schema.

robherring, 2024-03-04:
I started to add it, but then:

$ git grep stdin-path
arch/m68k/sun3/prom/console.c:          prom_getproperty(prom_root_node, "stdin-path", propb, sizeof(propb));

Yes, that's "just" Linux, but seems unlikely anywhere else will support it. So maybe it should just be removed from the spec instead? Do you have a usecase for it?

vstehle, 2024-07-05:
> So maybe it should just be removed from the spec instead?

Hi @robherring, thanks for having a look.
You are right, this sounds like the sensible thing to do.

kettenis, 2025-05-29:
The sun3 PROM predates Open Firmware. I don't think it really had the concept of a device tree. And that code is actually #ifdef 0'ed in the Linux kernel...

Open Firmware has both /chosen/stdout and /chosen/stdin, so without the -path suffix, since in Open Firmware you can directly reference a device tree node. But otherwise they address the same concept as /chosen/stdout-path and /chosen/stdin-path, allowing the firmware to designate the console output and input device. Which means you can do funny things like having output on a display, but input on serial.

Having a way to communicate the firmware's idea of the console device to the OS could be useful in cases where the OS considers the console device priviliged and you want to prevent someone from just plugging in a USB keyboard to make use of those priviliges. OpenBSD/sparc64 does use /chosen/stdin to determine the console input device. We probably would have used /chosen/stdin-path on devicetree based systems if it was used.

closed 2026-09-17 by robherring