devicetree-org/dt-schema #139: PCI binding: External interrupt-parent

FlyGoat, 2024-06-12:
I'm trying to get MIPS `loongson/loongson64g_4core_ls7a.dtb` passing dt schema checks.

After fixing #100 I fall into errors like:
```
/home/flygoat/linux-next/arch/mips/boot/dts/loongson/loongson64g_4core_ls7a.dtb: pci@1a000000: ohci@4,0:interrupts-extended:0: [4, 49, 4] is too long
        from schema $id: http://devicetree.org/schemas/pci/loongson.yaml#
```
On nodes like:
```
			ohci@4,0 {
				compatible = "pci0014,7a24.0",
						   "pci0014,7a24",
						   "pciclass0c0310",
						   "pciclass0c03";

				reg = <0x2000 0x0 0x0 0x0 0x0>;
				interrupts-extended = <&pic 49 IRQ_TYPE_LEVEL_HIGH>;
			};
```

This is because in our pci device schema `pci-device.yaml`, we assumed `interrupts` must be one of the PCI legacy interrupt:
```
  interrupts:
    items:
      - items:
          - enum: [1, 2, 3, 4]  # INTA, INTB, INTC, INTD
```

I do think `loongson64g_4core_ls7a.dtb`'s use case is somehow legitimate, how can we reasonably accommodate in schema?

Thanks

robherring, 2024-06-12:
You need an interrupt-map in the PCI host bridge to map the PCI interrupts to the &pic.

FlyGoat, 2024-06-12:
> You need an interrupt-map in the PCI host bridge to map the PCI interrupts to the &pic.

It's pretty hard to do it this way due to limitations on granularity of `interrupt-map`, our pci_bridges have different-ish mapping rule.

Also some devices have more than one interrupts from devicetree.

robherring, 2024-06-12:
So you are saying the PCI interrupt for the device has no effect? Or you have both PCI interrupts and some sideband interrupts?

FlyGoat, 2024-06-12:
> So you are saying the PCI interrupt for the device has no effect? Or you have both PCI interrupts and some sideband interrupts?

PCI interrupt from the device has no effect. Some of devices are not using standard PCI interrupt delivery. Only PCI to PCI bridges under the host controller is using standard PCI interrupt.

robherring, 2025-04-15:
I added 'interrupts-extended' to the PCI device schema so that sideband interrupts can be used. I think that should work in this case.

closed 2025-04-15 by robherring