devicetree-org/dt-schema #145: Devicetree node for the PCI slots

Mani-Sadhasivam, 2024-10-07:
Hi,

I'm trying to add a node for PCI(e) slots to represent the supplies correctly instead of adding them in host bridge node.

Like,

```
&pcieport0 {
        slot@0 {
                compatible = "pcie,slot";
                reg = <0x10000 0x0 0x0 0x0 0x0>;
                vpcie12v-supply = <&vpcie12v_reg>;
                vpcie3v3-supply = <&vpcie3v3_reg>;
                vpcie3v3aux-supply = <&vpcie3v3aux_reg>;
        };
};
```
But the supplies are already part of the `pci-bus-common.yaml`. Does it make sense to create a separate binding for the slot? We will need different compatibles to differentiate between slots like `pcie,mini-slot`, `pcie,m.2-slot` etc...

robherring, 2024-10-07:
I don't think we can do this because what child nodes are for PCI is already defined. How would this look if the PCI card node was actually populated? Our choice is do the properties go in the bridge node or device node. Given PCIe is point to point, it shouldn't really matter. PCI is where it would have mattered possibly, but do we care at this point?

Mani-Sadhasivam, 2024-10-08:
If we add these supplies to the bridge node, then it implies that the bridge node itself needs these supplies, but it doesn't. And we cannot describe device nodes because these are slots where the user can plug any devices/cards.

We do describe some devices connected to M.2 slots like WLAN modules from Qcom but they often require external supplies other than the PCIe spec defined ones.

robherring, 2024-10-08:
> If we add these supplies to the bridge node, then it implies that the bridge node itself needs these supplies, but it doesn't. 

Kind of. But child nodes are defined to be devices. We can't define different child nodes. That train has left the station.

> And we cannot describe device nodes because these are slots where the user can plug any devices/cards.

That's not true because we generate the nodes sometimes now. If a user plugs in an FPGA card and wants to describe devices within the FPGA, then you need the PCI device node.

> We do describe some devices connected to M.2 slots like WLAN modules from Qcom but they often require external supplies other than the PCIe spec defined ones.

Not a standard slot then.

robherring, 2024-10-10:
> So your suggestion is to add the supplies to the bridge nodes because we cannot predict the device node all the time. In that case, I think we need two types of nodes. If the slot is a standard one and the device doesn't need any external supplies, then the supplies need to be added to the bridge node. And if the device needs external supplies, then they go to device specific nodes.

Essentially, yes. Standard (slot) properties go in parent, non-standard in device node. Ideally, we'd put slot properties in the root port nodes, but we have to fall back to the host bridge nodes because some bindings already do that. The schema already covers all this for supplies at least.

Mani-Sadhasivam, 2024-10-10:
Thanks for clearing it up!

closed 2024-10-10 by Mani-Sadhasivam