devicetree-org/dt-schema #149: ValueError: unknown url type

JohnTheCoolingFan, 2024-12-27:
Working on patching linux kernel and getting an error `ValueError: unknown url type: 'cache-controller.yaml'`

The short commit hash of the linux source code is `d6ef8b40d075c` (latest master), and the full command is `CROSS_COMPILE=aarch64-unknown-linux-gnu- ARCH=arm64 make CHECK_DTBS=y rockchip/rk3566-bigtreetech-pi2.dtb rockchip/rk3566-bigtreetech-cb2-manta.dtb` (checking device trees added by a patch series).

`dtschema` version 2024.11 installed on Gentoo using ebuild from [peec overlay](https://github.com/rodan/overlay)

Full error message:

```
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/referencing/_core.py", line 423, in get_or_retrieve
    resource = registry._retrieve(uri)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/validators.py", line 112, in _warn_for_remote_retrieve
    request = Request(uri, headers=headers)  # noqa: S310
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/urllib/request.py", line 318, in __init__
    self.full_url = url
    ^^^^^^^^^^^^^
  File "/usr/lib/python3.12/urllib/request.py", line 344, in full_url
    self._parse()
  File "/usr/lib/python3.12/urllib/request.py", line 373, in _parse
    raise ValueError("unknown url type: %r" % self.full_url)
ValueError: unknown url type: 'cache-controller.yaml'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/referencing/_core.py", line 672, in lookup
    retrieved = self._registry.get_or_retrieve(uri)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/referencing/_core.py", line 430, in get_or_retrieve
    raise exceptions.Unretrievable(ref=uri) from error
referencing.exceptions.Unretrievable: 'cache-controller.yaml'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.12/dt-validate", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 154, in main
    sg.check_dtb(filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 91, in check_dtb
    self.check_subtree(dt, subtree, False, "/", "/", filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 84, in check_subtree
    self.check_subtree(tree, value, disabled, name, fullname + name, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 79, in check_subtree
    self.check_node(tree, subtree, disabled, nodename, fullname, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 34, in check_node
    for error in self.validator.iter_errors(node, filter=match_schema_file,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/dtschema/validator.py", line 456, in iter_errors
    for error in self.DtValidator(schema,
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/validators.py", line 384, in iter_errors
    for error in errors:
                 ^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_keywords.py", line 386, in if_
    yield from validator.descend(instance, then, schema_path="then")
  File "/usr/lib/python3.12/site-packages/jsonschema/validators.py", line 432, in descend
    for error in errors:
                 ^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 423, in unevaluatedProperties_draft2019
    evaluated_keys = find_evaluated_property_keys_by_schema(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 399, in find_evaluated_property_keys_by_schema
    evaluated_keys += find_evaluated_property_keys_by_schema(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 342, in find_evaluated_property_keys_by_schema
    resolved = validator._resolver.lookup(ref)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/referencing/_core.py", line 676, in lookup
    raise exceptions.Unresolvable(ref=ref) from error
referencing.exceptions.Unresolvable: cache-controller.yaml#
```

robherring, 2024-12-27:
You are using jsonschema 4.18 or newer which is not supported. See issue #109.

closed 2024-12-27 by robherring

JohnTheCoolingFan, 2024-12-27:
thanks, I will forward this to the overlay maintainer.