devicetree-org/dt-schema #151: Traceback on DTC

nunojsa, 2025-01-14:
When validating a binding, I'm getting the following trace:

```
Traceback (most recent call last):
  File "/home/nsa/work/linux-iio/.venv/bin/dt-validate", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb_validate.py", line 154, in main
    sg.check_dtb(filename)
    ~~~~~~~~~~~~^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb_validate.py", line 89, in check_dtb
    dt = self.validator.decode_dtb(f.read())
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/validator.py", line 555, in decode_dtb
    return [dtschema.dtb.fdt_unflatten(self, dtb)]
            ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb.py", line 518, in fdt_unflatten
    fixup_addresses(validator, dt, 2, 1)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb.py", line 490, in fixup_addresses
    fixup_addresses(validator, v, ac, sc)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb.py", line 490, in fixup_addresses
    fixup_addresses(validator, v, ac, sc)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb.py", line 490, in fixup_addresses
    fixup_addresses(validator, v, ac, sc)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nsa/work/linux-iio/.venv/lib/python3.13/site-packages/dtschema/dtb.py", line 495, in fixup_addresses
    while i < len(val):
              ~~~^^^^^
TypeError: object of type 'sized_int' has no len()
```

The behavior is surely new. I've tried to run dtschema also in docker for a cleaner environment but got the same result. Is this is a bug or am I missing something?

I also tried with the 2024.9 tag and I'm pretty sure I successfully validated bindings after September. Something odd is going on... Maybe an update in one of the dependencies?

robherring, 2025-01-14:
I may have seen this, but it shouldn't happen with current release and linux-next or Linus' trees. Can you narrow this down to what version of schemas and what dtb.

nunojsa, 2025-01-14:
At this point, I've tried with 2024.9, 2024.11 and main... My dtc version is:

```
dtc --version
Version: DTC v1.7.2-1-gac324a9
```

I could still not successfully validate any binding so I could still not find a valid combination that works for me... I'll get back to this tomorrow and try it in a different machine. Ah, and I also tried it on the dt tree (dt/next branch) so I'll make sure to try it on next or Linus's tree as well.

> but it shouldn't happen with current release

That was also my assumption, otherwise your bot should have been busy sending emails all over the place...

robherring, 2025-01-14:
You aren't giving me anything I can try to reproduce... 

BTW, your installed 'dtc' is not what is being used here. It's pylibfdt which is it's own copy installed via pip (unless you are manually doing something else).

nunojsa, 2025-01-15:
> You aren't giving me anything I can try to reproduce...

> BTW, your installed 'dtc' is not what is being used here. It's pylibfdt which is it's own copy installed via pip (unless you are manually doing something else).

Sorry for that... Anyways I tracked down the problem. It was related with the binding I was validating (in fact it was CI job first doing it but the output was odd enough). The problem was due to:

```
reg:
 description: Channel number.
 $ref: /schemas/types.yaml#/definitions/uint32
 minimum: 0
 maximum: 9
```

reg is standard and already has a type. What confused me was that having this in one file and the error would pop up for all the other bindings I tried to validate. Not sure if there's much you can do about this but if we can improve the error message in order to easily detect the problem, that would be great.

Since I have your attention, what linux tree would you recommend to use to run dt-bindings validations in a CI context?

Thanks!

robherring, 2025-01-15:
> Sorry for that... Anyways I tracked down the problem. It was related with the binding I was validating (in fact it was CI job first doing it but the output was odd enough). The problem was due to: [...]

The way it works is the type info is gathered for all properties. If a given property has multiple types, then we have to pick one when decoding the dtb. How well that works depends on the types. I'd like to warn, but we do have a few cases that we have to live with. 

I might be able to make the meta-schema ban $ref for 'reg'. 

> 
> Since I have your attention, what linux tree would you recommend to use to run dt-bindings validations in a CI context?

Depends what you want to test. I run next and Linus daily: https://gitlab.com/robherring/linux-dt/-/jobs Then I have a few scripts which process the logs in various ways such as new undocumented compatibles used in drivers or top warnings by each sub arch (strips the dtb name to deduplicate the warnings since 1 SoC warning occurs N times for N boards). I need to push those out somewhere...

robherring, 2025-01-15:
I just added a check for handling cases of multiple types where decoding them correctly is impossible.

BTW, You can run 'dt-extract-props -d Documentation/devicetree/bindings/' and get a list of all properties with multiple types.