devicetree-org/dt-schema #165: dt-schema complains one entry phandle-array is not of type 'array'.

inochisa, 2025-06-13:
binding link: https://lore.kernel.org/all/20250611081000.1187374-2-inochiama@gmail.com/
dts: link: https://lore.kernel.org/all/20250611082452.1218817-3-inochiama@gmail.com/

I got an compile when testing
```
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: syscon@3000000 (sophgo,cv1800b-top-syscon): dma-router@154:dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/soc/sophgo/sophgo,cv1800b-top-syscon.yaml#
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: syscon@3000000 (sophgo,cv1800b-top-syscon): dma-router@154:dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/soc/sophgo/sophgo,cv1800b-top-syscon.yaml#
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: dma-router@154 (sophgo,cv1800b-dmamux): dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/dma/sophgo,cv1800b-dmamux.yaml#
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: dma-router@154 (sophgo,cv1800b-dmamux): dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/dma/sophgo,cv1800b-dmamux.yaml#
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: dma-router@154 (sophgo,cv1800b-dmamux): dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/dma/dma-router.yaml#
linux-mainline/cv18xx/arch/riscv/boot/dts/sophgo/cv1800b-milkv-duo.dtb: dma-router@154 (sophgo,cv1800b-dmamux): dma-masters: 4 is not of type 'array'
	from schema $id: http://devicetree.org/schemas/dma/dma-router.yaml#
  DTC [C] arch/riscv/boot/dts/sophgo/cv1812h-huashan-pi.dtb
```

Is this an expected behavior?

krzk, 2025-06-13:
Seems yes, because you put invalid cells/arguments. Patch is incomplete to see that, so please always provide full context.

inochisa, 2025-06-13:
> Seems yes, because you put invalid cells/arguments. 
I am happy to know which one is invalid. I double check the dma-router.yaml and the dmamux binding I add in the first url and find nothing wrong.

IIRC, the `dma-masters` is a phandle-array with each phandle have one element (from dma-router.yaml), and the array should only have one items (from [my dmamux bindings](https://lore.kernel.org/all/20250611081000.1187374-2-inochiama@gmail.com/)), is this wrong for this device?

> Patch is incomplete to see that, so please always provide full context.
I have add some necessary information for this.

krzk, 2025-06-13:
Hm, now I think that maybe my comment on cells is not pointing out the issue. dma-masters don't follow cells because inner phandle size is always 1 (items: maxItems: 1), so indeed the hack could be the issue here.

inochisa, 2025-06-13:
> Hm, now I think that maybe my comment on cells is not pointing out the issue. dma-masters don't follow cells because inner phandle size is always 1 (items: maxItems: 1), so indeed the hack could be the issue here.

Yeah, I have decoded the dtb and find the `dma-masters` is assigned as 0x04.....

robherring, 2025-06-27:
I reworked my hack and made it a bit smarter. Now it checks if the value is a valid phandle and a node containing #dma-cells. Please give main branch a spin and let me know if it doesn't work for you.

closed 2025-06-27 by robherring